longzheng/homebridge-fronius-inverter-lights #9: Include Battery data

Druide85, 2022-06-19:
Thanks for the nice plug-in,

However, is it possible to include also the power flow to the battery or back from the battery. This would be a nice additional feature. 

Thanks a lot

longzheng, 2022-07-23:
Sorry I don't have a Fronius-compatiable battery system (I have a Tesla Powerwall) so I don't see any battery data in the API to test against.

Do you have any sample data at `http://[your-fronius-ip]/solar_api/v1/GetPowerFlowRealtimeData.fcgi`?

systemx-xx, 2022-08-26:
Battery SOC sits in Inverters.1.SOC

`{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "Battery_Mode" : "normal",
               "DT" : 1,
               "E_Day" : null,
               "E_Total" : 546947.51694444439,
               "E_Year" : null,
               "P" : 7327.86865234375,
               "SOC" : 100.0
            }
         },
         "Site" : {
            "BackupMode" : false,
            "BatteryStandby" : true,
            "E_Day" : null,
            "E_Total" : 546947.51694444439,
            "E_Year" : null,
            "Meter_Location" : "grid",
            "Mode" : "bidirectional",
            "P_Akku" : 2.0681488513946533,
            "P_Grid" : -3886.98,
            "P_Load" : -3440.88865234375,
            "P_PV" : 7529.881591796875,
            "rel_Autonomy" : 100.0,
            "rel_SelfConsumption" : 46.956199893719628
         },
         "Smartloads" : {
            "Ohmpilots" : {}
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-08-26T01:33:09+00:00"
   }
}`

longzheng, 2022-11-18:
@systemx-xx Hey mate, I saw you made a fork at https://github.com/systemx-xx/homebridge-fronius-inverter-lights and added the SOC support.

If you want to open a PR I'm more than happy to merge it back in.

systemx-xx, 2022-11-24:
hey @longzheng I have not had a chance to play around with that since I forked it as displaying the inverter data using lights etc is pretty sub par to be honest. I really wish Apple would open up the ability for custom displays.

longzheng, 2022-11-24:
Fair enough. I noticed the [homebridge-tesla-powerwall](https://github.com/datMaffin/homebridge-tesla-powerwall) plugin uses a fan speed to display energy flow which has the advantage of showing actual values in the overview.

[image]

I've been using that plugin since getting a Powerwall and have been pretty happy with it. I wonder if this plugin should also do the same.

systemx-xx, 2022-11-24:
Ah thats clever, yeah thats quite a bit nicer.

longzheng, 2023-06-24:
I've just published a new version 1.5.1 that adds a battery accessory if you enable it in the config.

```
        {
            "name": "Fronius inverter",
            "platform": "FroniusInverterLightsPlatform",
            ...
            "battery": true
        },
```

Since I don't have a battery connected to my Fronius inverter I can't test it myself so let me know if there's any issues.

closed 2023-06-24 by longzheng

Druide85, 2023-06-25:
Thanks a lot. It works well. Best regards Thomas Von meinem iPhone gesendet