longzheng/homebridge-fronius-inverter-lights #18: how is it possible to check battery-state?

Scoo27, 2023-11-15:
i have a BYD 5,1kWh Battery installed. How to chek the balance?

longzheng, 2024-02-06:
Sorry I don't have a BYD set up so I can't validate this, but if the BYD is appearing in your Fronius portal then it should appear if you add `"battery": true` to your config.

Gulaschsuppe1, 2024-02-19:
Hi there
I added a BYD Battery successully. The battery % is correct in the homebridge UI. However the battery accessory in Homekit is off by default. This results in Homekit not showing a %, but only that the light "battery" is off. Can the battery "light" be changed to show the value in Homekit as % or as a lux value?

longzheng, 2024-02-23:
> thanks, that worked for me, BUT the power-output and Input when charging have the same positive Value, so it is impossible to see whether the battery is charging or discharging. Can you add a "+" or "-" for that?

Ah unfortunately HomeKit doesn't support negative values for the lightbulb entity (which I'm using since there's no battery).

I think the only thing I could do is have 3 different lightbulbs for the battery
- battery % - brightness represents state of charge
- battery out - on if battery is discharging, lux as amount of power discharging
- battery in - on if battery is charging, lux as amount of power charging

Scoo27, 2024-02-23:
> I think the only thing I could do is have 3 different lightbulbs for the battery

Sounds good to me
would be great if you could implement that,

longzheng, 2024-02-26:
Sure but I've been a bit busy so not 100% sure when I can get around to it.

longzheng, 2024-03-08:
@Scoo27 I published a new version with the battery changes.

closed 2024-03-08 by longzheng