longzheng/homebridge-fronius-inverter-lights #21: Add instructions for enabling JSON API on newer inverters, and provide ability to customise the HomeKit Model and Serial

Shaun-R, 2024-06-26:
Searching the 'net for a Homebridge plugin for my Fronius PV inverter, fancy coming across a familiar name! I remember you wrote a certain iOS app to allow topping up a certain public transport smart card to solve a certain crappy mobile website issue, as I recall... :) 

Anyway, your plugin works brilliantly mate! Thank you! 

Newer Fronius inverters have a redesigned web interface. Before setting up your plugin, one must first login to the local web UI as _Customer_ or _Technician_ user, open the hamburger menu on the left, select **Communication**, select **Solar API**, then turn on **Activate communication via Solar API**, and finally select **Save**. Otherwise, the inverter does not make the JSON data available. 

[image]

It might be worth adding this to the readme.md? 

My feature request for this plugin would be to allow user customisation of the Serial Number and Model which appears in HomeKit (as seen in below screenshot)

[image]

It appears that _fronius-accessory.ts_ currently configures the Manufacturer and Model values (line 95-97), though I don't see references to a serial number in your code. Would it be possible to expose these as user-configurable fields in the config.schema.json and pass them through to this?

Thanks for your work!

longzheng, 2024-06-26:
Hi Shaun, nice seeing you again too.

I'm travelling interstate this week but those suggestions makes a lot of sense. Will do when I get back. 

For the model and serial, I wonder if the API contains any of that actually. Will investigate.

Shaun-R, 2024-06-26:
Safe travels!

longzheng, 2024-07-01:
@Shaun-R It looks like I can automatically get the inverter model(s) from the `http://.../solar_api/v1/GetInverterInfo.cgi` endpoint which returns a `DT` number.

```json
{
   "Body" : {
      "Data" : {
         "1" : {
            "DT" : 115,
            "UniqueID" : "1189726",
            ...
         }
      }
   },
   ...
}
```

Then I can lookup the number in `http://.../solar_api/v1/data/DeviceDB_Data.json`

```json
{
  "Inverters": {
    "42": {
      "ProductName": "Symo Advanced 10.0-3-M",
      "DeviceFamily": "0",
      "NominalPower": "10000",
      "PhaseCountAC": "1"
    },
    "43": {
      "ProductName": "Symo Advanced 20.0-3-M",
      "DeviceFamily": "28",
      "NominalPower": "20000",
      "PhaseCountAC": "3"
    },
    ...
  }
}
```

I'm thinking I can use the inverter `ProductName` as the model and `UniqueID` as the serial, but because one system can have multiple inverters (which I merge into one PV accessory), I guess they'll just need to be joined with an ampersand. Does that sound reasonable to you?

Shaun-R, 2024-07-02:
Nice investigative work! Joining multiple UniqueID’s in the format [UniqueID][space][ampersand][space][UniqueID] seems quite reasonable to me!

longzheng, 2024-07-06:
@Shaun-R I started working on this but then I ran into a major problem, HomeKit/Homebridge seems to only initialize the accessory information (manufacturer, model, serial) only once when the accessory is loaded, and it can't be dynamically updated.

See https://stackoverflow.com/questions/75571300/homebridge-update-accessoryinformation and https://github.com/homebridge/homebridge/issues/1643

The only workaround I can think of is changing how the plugin platform initially creates the accessories, and introduce a delay there to wait for the API query to finish, then create the accessories with the metadata already loaded. I think that'll technically work, but the downside is that it will delay the "bridge startup" which isn't a big deal but is advised against.

longzheng, 2024-07-06:
@Shaun-R I've published a new beta version with the above changes, can you test

[image]

Shaun-R, 2024-07-24:
Hey mate, thanks for your work and update on this. 
For posterity, here is my config.json: 
```json
{
            "inverterIp": "192.168.1.16",
            "pollInterval": 10,
            "pvMaxPower": 10000,
            "battery": false,
            "platform": "FroniusInverterLightsPlatform"
},
```

Unfortunately I am getting error messages in my Homebridge logs: 
```
[7/23/2024, 1:47:59 PM] [FroniusInverterLightsPlatform] Request failed with status code 404
[7/23/2024, 1:47:59 PM] Initializing platform accessory 'Import'...
[7/23/2024, 1:47:59 PM] Initializing platform accessory 'Export'...
[7/23/2024, 1:47:59 PM] Initializing platform accessory 'Load'...
[7/23/2024, 1:47:59 PM] Initializing platform accessory 'PV'...
```
and later on: 
```
[7/23/2024, 10:27:51 PM] [FroniusInverterLightsPlatform] connect ENETUNREACH 192.168.1.16:80 - Local (0.0.0.0:0)
[7/23/2024, 10:27:51 PM] [FroniusInverterLightsPlatform] connect ENETUNREACH 192.168.1.16:80 - Local (0.0.0.0:0)
[7/23/2024, 10:27:51 PM] [FroniusInverterLightsPlatform] connect ENETUNREACH 192.168.1.16:80 - Local (0.0.0.0:0)
[7/23/2024, 10:27:51 PM] [FroniusInverterLightsPlatform] connect ENETUNREACH 192.168.1.16:80 - Local (0.0.0.0:0)
```

I can `ping 192.168.1.16` from the Homebridge Terminal just fine. 
I can reach `http://192.168.1.16/solar_api/v1/GetInverterInfo.cgi` just fine
However, `http://192.168.1.16/solar_api/v1/data/DeviceDB_Data.json` returns a 404 Not Found

I'm not sure where you got the information regarding `http://.../solar_api/v1/data/DeviceDB_Data.json`, as I can't find reference to this call in the [Fronius Solar API v1 documentation](https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2012.pdf). 

However, I think `http://.../solar_api/v1/GetActiveDeviceInfo.cgi` may be helpful as it returns a `DT` DeviceType value and a `Serial` value  for each Inverter, Meter, Ohmpilot, Sensor Card, and Storage device connected. This is outlined on page 39 of the API documentation, and Page 82 of the documentation then provides an Inverter Device Type List, which you could look up against. 

Alternatively, `http://.../solar_api/v1/GetPowerFlowRealtimeData.fcgi` also returns the Inverter DT (DeviceType) value but without the `Serial` value. 

Finally, an "easy way out" in my opinion is via `http://.../solar_api/v1/GetInverterInfo.cgi`, which returns a `CustomName` field which could be used for the Model field, and a `PVPower` field which you could use to obviate the need to manually specify a `pvMaxPower` in the config.json. Whilst the Site Name is not exactly a Model number, I think it's a case of "close enough is good enough"?

longzheng, 2024-07-24:
@Shaun-R I've published a beta 2 with a hard-coded device list (removed `DeviceDB_Data.json`) lookup, and also automatically calculating the `pvMaxPower` from the `GetInverterInfo.cgi`. Let me know how that goes.

(I found the `DeviceDB_Data.json` by looking at the web UI directly, it wasn't documented in their docs).

Shaun-R, 2024-07-25:
Update installed - looks good! 
[image]
I've removed
```json
"pvMaxPower": 10000,
```
from my `config.json` and this appears to be functioning correctly, too. 
Nice work!! :)

longzheng, 2024-07-25:
Cool. I'll publish this.

closed 2024-07-25 by longzheng