longzheng/homebridge-fronius-inverter-lights #24: Export bulb not accurate

oandpg, 2024-08-10:
Hi Love the plugin, I am using it to alert me when I am importing or exporting electricity so i can modify my usage. One problem is that the Export Light Bulb is not showing as “on” even when the lux value is >0 and I am exporting.
It seems work only when the %value on the switch is below 100.  The inverter is capable of exceeding its watt value by at least 2x 
Not sure if the other bulbs have the same issue.

Thanks, Ros

[image]
[image]

oandpg, 2024-08-10:
Also not sure how the % is calculated, but my system is 8.5kw approx. so 1500 is nowhere near 100%

longzheng, 2024-08-10:
@oandpg Hey. I made some changes recently around how the "max PV power" is calculated and maybe there's a bug.

Can you help me investigate what your inverter API values are. Can you copy and paste the output from

```
http://[inverterIp]/solar_api/v1/GetInverterInfo.cgi
http://[inverterIp]/solar_api/v1/GetPowerFlowRealtimeData.fcgi
```

oandpg, 2024-08-10:
{
   "Body" : {
      "Data" : {
         "1" : {
            "CustomName" : "not my real name",
            "DT" : 1,
            "ErrorCode" : 0,
            "InverterState" : "Running",
            "PVPower" : 6580,
            "Show" : 1,
            "StatusCode" : 7,
            "UniqueID" : "34775912"
         }
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-08-10T05:22:24+00:00"
   }
}

oandpg, 2024-08-10:
{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 1,
               "E_Day" : null,
               "E_Total" : 951372.34055555554,
               "E_Year" : null,
               "P" : 1936.397216796875
            }
         },
         "SecondaryMeters" : {},
         "Site" : {
            "BackupMode" : false,
            "E_Day" : null,
            "E_Total" : 951372.34055555554,
            "E_Year" : null,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -2426.8699999999999,
            "P_Load" : 491.08777343749989,
            "P_PV" : 2031.4679870605469,
            "rel_Autonomy" : 100.0,
            "rel_SelfConsumption" : 0.0
         },
         "Smartloads" : {
            "Ohmpilots" : {}
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-08-10T05:24:39+00:00"
}
}

oandpg, 2024-08-10:
I have two inverters 1.75 and 6.6. The fronius meter handles both.vI am  querying the larger 6.6 one. I am really only interested in the lux figures when >0 for my use case.

longzheng, 2024-08-10:
OK. Just confirming, in your first post, the two screenshots are from the same inverter right?

Looking at your output data, it all looks fine and I would have expected it to behave correctly.

The code that determines is the export bulb is on and what the brightness is here https://github.com/longzheng/homebridge-fronius-inverter-lights/blob/556bda129409d8e95b54bad93bae1ea8d148cf13/src/fronius-accessory.ts#L169-L174

It should all be working correctly since your `"rel_SelfConsumption"` is `0.0`

oandpg, 2024-08-10:
Yes same inverter.

longzheng, 2024-08-10:
Just to rule some other stuff out, do you mind having a look at your Homebridge Accessories page (you might need to enable "Insecure" mode), does the "Export" bulb have the same issue there?

Also there's no errors or anything in your Homebridge logs?

oandpg, 2024-08-10:
Yes insecure mode is on in Homebridge and accessory has the same issue in Homebridge.
[image]

longzheng, 2024-08-10:
That's super weird. I'd love to debug this with you on a video call. Can you email me at long.zheng@gmail.com

oandpg, 2024-08-10:
Also tried with the prev 1.62 version of the plugin and same issue. So not your recent changes

longzheng, 2024-08-10:
Are all your other bulbs (import, PV, load) working correctly?

oandpg, 2024-08-10:
Yes they seem OK.

longzheng, 2024-08-10:
Can you try press the little cog on the export bulb and then screenshot the detailed modal window?

[image]

oandpg, 2024-08-10:
There is no cog on mine. Is there a homebridge setting?

longzheng, 2024-08-10:
I see it when I hover over the accessory card.

oandpg, 2024-08-10:
[image]

The joys of no hover on an iPad… 😂

oandpg, 2024-08-10:
OK It has fixed itself. Some kind of odd caching problem I think. I rebooted the Raspberry iPi and restarted everything and it is now working.

oandpg, 2024-08-10:
Thanks for your time and help and sorry to be a pest.

longzheng, 2024-08-10:
No worries. It's odd since the code that updates the light sensor lux is the same code to update the bulb, so I don't see why one would update and the other would not.

Before you restarted the Raspberry Pi did you also try restart Homebridge itself?

oandpg, 2024-08-10:
Yes, many times.

longzheng, 2024-08-10:
Right. Super weird then.

oandpg, 2024-08-10:
🤣

oandpg, 2024-08-10:
May not be working. I realised that the % is bwlow 100 and it seems to work then, the blub goes off OVER 100%.  Will see tomorrow with more light…

longzheng, 2024-08-10:
Just triple checking, is it possible the export and import got renamed incorrectly?

oandpg, 2024-08-10:
No because the lux values match the fronius solar web values.

oandpg, 2024-08-10:
I am wondering if the inverter returns values that are weird. I will check tomorrow.

oandpg, 2024-08-11:
Good morning. I can confirm it does not work *ie the export bulb turns off) if the inverter returns 

"rel_Autonomy" : 100.0,
"rel_SelfConsumption" : 0.0

longzheng, 2024-08-11:
@oandpg I'll investigate this further. I need to add some tests to this project so I can mock data.

longzheng, 2024-08-11:
@oandpg Ah sorry I had a very obvious bug that I didn't notice before. https://github.com/longzheng/homebridge-fronius-inverter-lights/blob/556bda129409d8e95b54bad93bae1ea8d148cf13/src/fronius-accessory.ts#L166

`data.Site.rel_SelfConsumption || 100` means that if `rel_SelfConsumption` is `0` then it becomes `100`.

I'm just publishing a fix now.

oandpg, 2024-08-11:
Great news, glad you found it. I am a retired developer and I *know* how you can stare at something for ages and not see it! Lol.

longzheng, 2024-08-11:
I've published version 1.7.1, which should fix it.

Note to self, fixed in https://github.com/longzheng/homebridge-fronius-inverter-lights/commit/30e449ac1cf73381cbb2417ca0956294d1e2a065

oandpg, 2024-08-11:
Thanks, I can confirm it is working perfectly now.

closed 2024-08-11 by oandpg