longzheng/homebridge-fronius-inverter-lights #25: Fronius Inverter does not start

na-ma-ste, 2024-08-28:
HI, 

since today (Homebridge ui 4.57.1; node 20.11.0, homebirdge 1.8.4) I got this Error meessage on the console and the fronius plugin does not start and causes homebridge to stop work:

AxiosError: timeout of 2000ms exceeded
    at RedirectableRequest.handleRequestTimeout (/var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/node_modules/axios/lib/adapters/http.js:657:16)
    at RedirectableRequest.emit (node:events:518:28)
    at RedirectableRequest.emit (node:domain:488:12)
    at Timeout.<anonymous> (/var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/node_modules/follow-redirects/index.js:210:12)
    at listOnTimeout (node:internal/timers:573:17)
    at processTimers (node:internal/timers:514:7)
    at Axios.request (/var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/node_modules/axios/lib/core/Axios.js:45:41)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:540:9)
    at processTimers (node:internal/timers:514:7)
    at FroniusApi.getInverterInfo (/var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/src/fronius-api.ts:34:13)
    at FroniusInverterLightsStaticPlatform.getDeviceMetadata (/var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/src/fronius-platform.ts:129:27)
    at /var/lib/homebridge/node_modules/homebridge-fronius-inverter-lights/src/fronius-platform.ts:48:30

longzheng, 2024-08-29:
From the error it looks like there's a problem connecting to your inverter. In the past when I've run into similar problems it's either due to some WiFi connectivity issue (rebooting the router) or a bug with the inverter (reboot the inverter/datamanager).

I think you should also investigate changing the plugin to a child bridge so this error does not break your entire Homebridge instance. You can find the documentation for that here https://github.com/homebridge/homebridge/wiki/Child-Bridges

na-ma-ste, 2024-08-29:
Hi @longzheng ,

thank you. i have now done a complete reboot of the PI and that has helped. The plugin is working again.

However, it always shows 100 (in the homebridge). Although other values are present.

Any Ideas?

longzheng, 2024-08-29:
Which light is showing the 100? Import or export?

na-ma-ste, 2024-08-29:
Import: 100
PV: 100
Export: 0
Load: 0

Values are continuous, i.e. no change for hours

Real Values:
Import: 1
PV: 80
Export: 50
Load: 100

longzheng, 2024-08-29:
@na-ma-ste That's very weird. Where abouts are you seeing the "real values"? 

Does the data in `http://YOUR_INVERTER_IP/solar_api/v1/GetPowerFlowRealtimeData.fcgi` for `P_Grid` `P_Load` `P_PV` look reasonable?

If so, can you try restarting Homebridge again?

na-ma-ste, 2024-08-29:
That works but i see some null values in the response:

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 232,
               "E_Day" : 63930,
               "E_Total" : 24200800,
               "E_Year" : 10403875,
               "P" : 4403
            }
         },
         "Site" : {
            "E_Day" : 63930,
            "E_Total" : 24200800,
            "E_Year" : 10403875,
            "Meter_Location" : "unknown",
            "Mode" : "vague-meter",
            "P_Akku" : null,
            "P_Grid" : null,
            "P_Load" : null,
            "P_PV" : 4403,
            "rel_Autonomy" : null,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-08-29T17:44:08+02:00"
   }
}

longzheng, 2024-08-29:
Hmm it looks like there's a fault with your smart meter connection, it's not sending the load value.

Can you try restart the Fronius inverter or data manager?

na-ma-ste, 2024-08-30:
Hi @longzheng ,

everything is available and correct in the SOlar Web from Fronius and on my Varta storage system (where I read the values from above).

I also had the restart idea yesterday. I'll give it a try.

I've also noticed that I'm operating my Fronius in ‘night mode’. This means that it is off from sunset and does not provide any data. So far this has not been a problem, but after updating to UI 4.57.1 I get an Error (will try to send the error, made a copy of the log yesterday but forgot to save it :/ )

longzheng, 2024-08-30:
> everything is available and correct in the SOlar Web from Fronius and on my Varta storage system (where I read the values from above).

Right, I believe SolarWeb gets values pushed from the inverter by other means (not using this Solar API which this plugin uses) and the Varta storage system I presume would have meters/CT clamps separately to the Fronius system.

I'm guessing if you visited your Fronius datamanager IP directly and viewed the local UI it would also not show the load/grid values (since it uses the Solar API).

na-ma-ste, 2024-09-02:
Hi @longzheng ,

i have done the restart, but the output looks same:

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 232,
               "E_Day" : 18224,
               "E_Total" : 24416302,
               "E_Year" : 10619366,
               "P" : 2265
            }
         },
         "Site" : {
            "E_Day" : 18224,
            "E_Total" : 24416302,
            "E_Year" : 10619366,
            "Meter_Location" : "unknown",
            "Mode" : "vague-meter",
            "P_Akku" : null,
            "P_Grid" : null,
            "P_Load" : null,
            "P_PV" : 2265,
            "rel_Autonomy" : null,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-09-02T13:00:26+02:00"
   }
}

longzheng, 2024-09-03:
@na-ma-ste I'm not sure what's happening, I've not seen this before but it looks like there's something wrong with the meter data.

Unfortunately this is a bit of roadblock if the API isn't able to provide the correct site/load data then it won't be able to display the correct numbers.

longzheng, 2024-09-03:
@na-ma-ste i saw someone write about this elsewhere.

Can you check the SolarWeb settings > components. Under Energy Meters look for column “Visible”. Make sure it is enabled.

na-ma-ste, 2024-09-04:
Hi @longzheng ,

I check the Fronius Webinterface. I have only these two elements that fit to your request (the other are network, password and so on):

[image]
[image]

It looks that everything is active. Or do i need to check something else?

longzheng, 2024-09-04:
I believe the other person was referring to the setting in Fronius Solarweb https://www.solarweb.com/

[image]

na-ma-ste, 2024-09-04:
Ok, in Solor web it looks like this:

[image]

longzheng, 2024-09-04:
@na-ma-ste Yeah it looks like there's something technically wrong with your meter, maybe a hardware fault? I would recommend contacting your solar installer to troubleshoot.

closed 2024-09-06 by longzheng