longzheng/homebridge-fronius-inverter-lights #27: Plugin crashes Homebridge if the inverter is not connected to internet.

oandpg, 2024-11-28:
Hi the plugin will crash homebridge and put it into a restart loop when the plugin cannot contact the inverter. Occasionally my fronius gen24 loses the internet connection and it needs restarting, but in the meantime homebridge crashes repeatedly. This is usually how I find out the inverter is offline, because all the Homebridge items stop working 

[image]

longzheng, 2024-11-28:
Ah interesting. I can look into that but in the interim I would recommend creating a child bridge for this plugin to prevent it crashing.

oandpg, 2024-11-28:
Works as a child plugin and doesn’t cause Homebridge to crash. I can live with this if it is unfixable. Thanks for the quick response.

closed 2024-11-30 by longzheng