longzheng/homebridge-fronius-inverter-lights #28: Deprecated Code

oandpg, 2025-01-12:
[image]

longzheng, 2025-01-12:
It looks like [`punycode` depedency comes from ESLint](https://github.com/eslint/eslint/issues/17720) which is used as a linting tool so it's not run in the normal runtime, I would consider this a low priority.

I'm a bit busy these days but I would probably prioritise upgrading to this plugin to Homebridge 2.0 template at some point but I believe this plugin will continue to work anyway https://github.com/homebridge/homebridge-plugin-template

oandpg, 2025-01-12:
No problem. I just thought I’d report it. Btw loving the plugin, I  use it to turn on appliances when there is a decent amount of export and off when they import. Agree that 2.0 is more important Thanks and cheersRosSent from my iPhone